HERA-Team/hera_mc #1: Uniqueness for part-port pairs.

dannyjacobs, 2016-10-26:
In the connections table the part+port need to be unique for both sides: A and B. (ddb_start branch)

miguelfmorales, 2016-10-26:
Just to make sure everyone is on the same page. A nice dB design pattern:  For every part there should be a table that has a unique id (key for the table) and a bunch of columns for any info you want about that part. (hardware tracking table) Then you have a linking table that contains two foreign table keys, and the time range when this connection was true. The foreign keys guarantee you can find the part, the time range lets you ask what was connected at date X. This means the part and port cannot be unique, as they may be reused (antenna 17 was plugged into receiver node 7 slot 1, then node 3 slot 8 for a while, then plugged back into node 7 slot 1).

dannyjacobs, 2016-10-26:
Good point, we agree.  Connections are not unique because of reconnects.  How should we make sure that every connection is only connected at one time?

miguelfmorales, 2016-10-27:
Not ideal, but it can be as part of the API and/or web interface used to insert the new connection. Either a check to make sure parts are not connected at the time (current if changing now, but could be retrospective) and/or for a 'change connection now' page something that automatically ends a connection if currently active and moves it to a new one. We use both of these patterns in the MWA scheduling. 

Just one idea.

closed 2016-11-12 by david-deboer