HERA-Team/hera_mc #3: Contents tables

dannyjacobs, 2016-10-26:
contents tables map where things are within subsystems. They mainly provide physical mappings within racks and shelves.

Node contents
rcvrator contents
possible: fengine rack contents

bhazelton, 2017-12-15:
@dannyjacobs Is this met by the cm stuff?

david-deboer, 2017-12-15:
I think our current daily ops plots meet this.

dannyjacobs, 2017-12-15:
> Assigned #3 <https://github.com/HERA-Team/hera_mc/issues/3> to

I don't know what this is either.  Did I open it?

bhazelton, 2017-12-15:
Yes, you did, and you wrote a comment that I think we've met.

dannyjacobs, 2017-12-15:
Ok, this is basically @adampbeardsley's plots. We'll need to update for the new system, but we close this one and open a new one in a set of all the stuff we'll need for that upgrade.

closed 2017-12-15 by dannyjacobs