HERA-Team/hera_mc #16: mc.py data path marginal

david-deboer, 2017-01-21:
mc.py defined a path to a data directory (hera_mc/data), which was a merge conflict.

The issue is that dirname(__file__) expands to (often, but not always) where hera_mc is installed in bin (or under some egg or another).  Since you write to this directory, it won't work in a repo.  The version that I did assumed that you were in the correct base directory, also not a good solution unless you made sure you were in the right directory to start...

So, how does one always find the place where the repo was git'ed, not where python installed?

I've put it back to my version, since at least that works for a defined case.  This doesn't matter unless you write to the data_path.

pkgw, 2017-01-23:
You can't reliably find the original source repository — it may not have ever existed, if you installed the software through a package manager. Or it could simply have been deleted after the software was installed.

It's also the case that you should never expect to be able to write files where your software was installed — i.e., you should treat those places as read-only.

So, I think the solution is not to try to write files to `data_path`. What's the use case for that? If you want to update the various text versions of the databases, I think the correct solution is to write them to the current directory and make it the user's responsibility to update the repository, or not.

david-deboer, 2017-01-24:
I accede to pkgw's logic and have changed accordingly.

closed 2017-01-24 by david-deboer