HERA-Team/hera_mc #29: use args_test code in scripts rather than repeated argument parsing code

bhazelton, 2017-05-31:


pkgw, 2017-06-17:
@david-deboer I am confused about the file `hera_mc/args_test.py`. If I do a `git grep` I just don't see `args_test` being used anywhere in the code. Can you clarify what the purpose of this file is?

david-deboer, 2017-06-17:
If I want to test aspects of code from within ipython you need an args generator.  So, it is only useful to set up an args there.  I haven’t used it much, but it is helpful to have around.

So, it is an args-for-interactive-testing, not a test-for-args

david-deboer, 2017-06-17:
I can annotate the file, but I'd suggest we keep it and close the issue.

pkgw, 2017-07-26:
Renamed to `args4testing.py`, purpose explained somewhat in a header comment.

closed 2017-07-26 by pkgw