HERA-Team/hera_mc #38: Deploy case-insensitive collation for CM databases, or work around inability to do so

pkgw, 2017-06-17:
The version of Postgres that we're using *should* make it possible to set a policy such that text searches on the CM database are case-insensitive by default, which would be nice to have since users might capitalize various parameters that they input to the CM system. Thus far, however, @david-deboer has been unable to get this working.

If we determine that there's some reason that really makes this impossible to deploy, we should come up with some scheme that makes things simple and reliable for users. For example, we could adopt a policy that all text items in the database are written without capital letters, always.

bhazelton, 2017-08-22:
@david-deboer Should this issue still be open?

david-deboer, 2017-08-22:
This was also fixed today.

closed 2017-08-22 by david-deboer