HERA-Team/hera_mc #47: Version tracking of CM database

plaplant, 2017-07-14:
For reproducibility of the CM database at a later date (e.g., recreating "what we thought the array looked like" in light of a change made to reflect "what it actually looked like"), we should have a way to refer to the exact state of the database at a specific point in history. For example, this could be achieved by making a `git commit` of the CSV files every time a new connection is entered into the database, and then optionally returning the hash of the commit when the database is queried. The goal (from an RTP point of view) is to be able to recreate the calfile used for calibration.

miguelfmorales, 2017-07-14:
I believe this is already handled. The way the database is designed it already tracks the historical state of the system (one of the main design drivers), so this is trying to solve a solved problem. I'll let someone closer confirm and close?

bhazelton, 2017-07-14:
@miguelfmorales you missed this portion of the discussion. What we're talking about here is tracking the state of the database, not the system. So in some cases, the wrong antenna position information or connection information is entered. Then it's corrected later, so that the database is an accurate representation of history. But any code run in the meantime needs to be able to be identified as needing to be re-run. So we need a way to actually version the database. This is really only an issue with the CM stuff, which is inherently subject to human error.

david-deboer, 2017-08-03:
The csv files in a repo should serve this function.  This will be done (committed) on any change and we'll make a daily cronjob. The csv files are now moved to another repo (hera_cm_db_updates).  Since the cronjob isn't set we can't close this.  The cronjob should
1 - cm_pack.py
2 - copy to appropriate directory
3 - git commit

pkgw, 2017-08-04:
The cronjob script should go in [HERA_Commissioning](https://github.com/HERA-Team/HERA_Commissioning/), which houses these sorts of scripts.

bhazelton, 2017-08-22:
@david-deboer things have shifted a bit with PRs #107 and #110. What do we still need to do to close this issue?

david-deboer, 2017-08-22:
I think that we can call this one closed with the script today.

closed 2017-08-22 by david-deboer