HERA-Team/hera_mc #57: According to M&C, antenna 6 active during H19 but its not in any of our data.

dannyjacobs, 2017-07-26:
[image]
According to M&C, Antenna 6 (HH51) was turned on and connected same time as the rest of H19, but we don't have it in our data. I've had to explicitly exclude it from CM integration tests over at hera_cal (because its not in our test data file), but would be nice to not have to do that.

bhazelton, 2017-10-25:
Has this been resolved?

bhazelton, 2018-07-27:
@dannyjacobs @david-deboer this issue is very old. Has it been resolved?

pkgw, 2018-08-14:
Gonna close this due to long-term inactivity. Can always reopen if needed.

closed 2018-08-14 by pkgw

dannyjacobs, 2018-08-14:
Ugh H0C?  Lets forget this ever happened.