HERA-Team/hera_mc #92: Need method to test connection to db

adampbeardsley, 2017-08-10:
Currently the method `connect_to_mc_db` will return happily, even if the database is unavailable/unreachable/nonexistent. We need a controlled way to test whether the connection was made and is active. Perhaps `connect_to_mc_db` should raise an exception or return a certain way if the connection is unsuccessful.

pkgw, 2017-08-10:
Which database backend are you using? I certainly get errors if my local Postgresql server isn't running.

adampbeardsley, 2017-08-10:
Hmm. I'm running the Postgres Mac app (https://postgresapp.com/). I've tried both dropping the db via psql prompt, as well as stopping the server altogether. In latter case, here is what I see:
```
(HERA) macbook-pro-2:hera_mc abeardsley$ python scripts/add_qm_metrics.py --type firstcal hera_mc/data/test_data/example_firstcal_metrics.json 
Traceback (most recent call last):
  File "scripts/add_qm_metrics.py", line 32, in <module>
    session.ingest_metrics_file(f, args.type)
  File "/Users/abeardsley/code/hera_mc/hera_mc/mc_session.py", line 954, in ingest_metrics_file
    obsid = self.get_lib_files(filename=os.path.basename(filename))[0].obsid
  File "/Users/abeardsley/code/hera_mc/hera_mc/mc_session.py", line 546, in get_lib_files
    LibFiles.filename == filename).all()
<... long traceback... can paste if needed...>
sqlalchemy.exc.OperationalError: (psycopg2.OperationalError) could not connect to server: Connection refused
	Is the server running on host "localhost" (::1) and accepting
	TCP/IP connections on port 5432?
could not connect to server: Connection refused
	Is the server running on host "localhost" (127.0.0.1) and accepting
	TCP/IP connections on port 5432?
```
In the former case (when dropping the db but keep the server running), the exception is at the same point in our code, but the last line of the traceback reads:
```sqlalchemy.exc.OperationalError: (psycopg2.OperationalError) FATAL:  database "hera_mc" does not exist```

What error do you see?

pkgw, 2017-08-10:
I get the same thing, but within the connect-to-database call. You could definitely imagine that different variations of SqlAlchemy might decide to be lazy about when they actually implement the database connection.

If we had `connect_to_mc_db` call `mc_check.is_sane_database` you'd probably catch errors when you want ...

adampbeardsley, 2017-08-10:
Ok, just talked to Paul for a bit on what he's looking for from the RTP standpoint. The solution seems to be that at the end of `connect_to_mc_db`, we call `db_check.is_sane_database`, like Peter suggested. We will have that in a `if not` statement so we can raise an exception if it returned false. That way RTP will always get an exception if something is wrong. If that makes sense to people, I will go ahead and start a branch to implement it.

pkgw, 2017-08-11:
Yeah, I think that's a good approach, especially since our vision was the the M&C systems would refuse to run if the expected and actual DB schemas differed.

I think our testing code uses the same function to connect to empty testing databases, though, so you might need to add a keyword argument that lets the caller override the "sanity" check if they're sure that they don't want it to happen.

(We might consider changing the name of the function since it is a little implicitly stigmatizing of mental illness.)

adampbeardsley, 2017-08-11:
I might need someone more familiar with the M&C schema to help me with this. It looks like the production mode database already does the `is_sane_database` check via the `AutomappedDB` class. So perhaps adding at the end of `connect_to_mc_db` is not necessary and in fact not the right thing to do because it should only be run when you want the full M&C db, which is when it will be called from `AutomappedDB.__init__`. Am I reading this correctly?

I agree we should change the name of the function. I propose `is_valid_database`.

closed 2017-08-12 by adampbeardsley