HERA-Team/hera_mc #142: Nightly RTP status update.  Number of files per night vs time, errors per night vs time?

dannyjacobs, 2017-09-20:
example code at hera_mc.autocorrelations.  @pkgw can help

dannyjacobs, 2017-09-26:
written but not tested on site.  Next: how to integrate with plotly?

plaplant, 2017-09-26:
Right now, the current output is to screen, and looks something like:

```
RTP report for JD 2458017 (Wed Sep 20, 2017)

Number of observations: 276
Number finished processing: 245 (89%)
Number currently processing: 27 (10%)
Number failed: 4 (1%)
```
The plan is to send as an email somewhere, though it'd be easy enough to tack this on to an existing summary script. I suppose another thing that could be added is a plot with JD on the x-axis, and status on the y-axis, so we can see if there are particular times that are still processing/failed. It might be possible to integrate that plot with the plotly report that gets generated, since this script is very lightweight.

dannyjacobs, 2017-10-31:
Done!

closed 2017-10-31 by dannyjacobs