HERA-Team/hera_mc #145: Add RFI plots

dannyjacobs, 2017-09-20:
"nightly max-hold spectrum" with known things labeled

It can show up on a server somewhere as a png, or can use plotly. examples can be found at hera_mc.autocorrelations

Can grab data from nrao librarian or run as cronjob.

dannyjacobs, 2017-09-26:
UW will work on refining the report.  
@adampbeardsley will get Mike an example OR file and the different channels of RFI flagging
Visualization needs to include an auto correlation showing what the data look like.

jaguirre, 2017-09-28:
Saul did so much much work in making nice RFI waterfalls of occupation time.  Surely this is re-useable?

dannyjacobs, 2017-09-28:
Mostly we need people next to this. Saul's already committed to so much. Waterfalls are pretty easy the hard part that we should capture from him is the list of bogeys.

jaguirre, 2017-09-28:
There's a historical list of offenders in the memos, but do you mean regenerating that from new(er) data?

dannyjacobs, 2017-09-28:
I mean, starting from that list and then adding or monitoring.

SaulAryehKohn, 2017-09-28:
The list(s) are [here](http://reionization.org/wp-content/uploads/2013/03/HERAMemo19_HERA_dish_RFI.pdf).

And the [plotting script for the waterfalls](https://github.com/HERA-Team/hera_sandbox/blob/master/sak/scripts/flagreader.py ) can also push-out npzs with occupation per channel/time data:

dannyjacobs, 2017-10-31:
Coming in next RTP update.

plaplant, 2017-12-15:
See Issue #84 on HERA_Comissioning (https://github.com/HERA-Team/HERA_Commissioning/issues/84) for further discussion.

closed 2017-12-15 by plaplant