HERA-Team/hera_mc #155: think up a psql-free API to plot against cm columns

dannyjacobs, 2017-10-17:


pkgw, 2017-10-17:
You can create in-memory SQLite databases, so you could imagine a single function that you can call that takes just one argument, the path to a checkout of the `hera_cm_db_updates` database, and gives you an instantiated SQLAlchemy session that points to an initialized database that you can query.

(The "in-memory" part doesn't buy you a ton except then you don't have to come up with a path to the SQLite file, and you don't have to worry about synching that file with the `hera_cm_db_updates` checkout. I'm assuming that ingesting the CSV files into the in-memory database is fast.)

pkgw, 2017-10-28:
Going to close this as a duplicate of #25 unless I'm missing something here.

closed 2017-10-28 by pkgw