HERA-Team/hera_mc #156: be able to tag data as "science" vs "engineering"

david-deboer, 2017-10-17:


jack-h, 2018-05-03:
I'm going to add a field which ends up in the data files for an observation tag (

jack-h, 2018-05-03:
Wow. Mega ineffective github-ing.

I'm going to add to the correlator interface an option to tag data files with a string when you start the correlator, to at least give some minimal record of what the observer's intention was.

bhazelton, 2019-01-18:
This is implemented in M&C. @jack-h does this tag go into the file header already?

jack-h, 2019-01-19:
This is implemented in the files as of fab0dadf (i.e., Just Now).

The missing link is the correlator actually responding to MC. Working on that now....

bhazelton, 2019-09-09:
@jack-h can this be closed? Are these tags written into the files?

jack-h, 2019-09-11:
Yes. Close at leisure.

closed 2019-09-11 by bhazelton