HERA-Team/hera_mc #157: implement an off-site mirror

dannyjacobs, 2017-10-19:
this will be necessary for data releases larger than a ~month

pkgw, 2017-10-19:
Can you edit the description to be more ... descriptive? I don't understand the issue and/or what it has to do with M&C.

dannyjacobs, 2017-10-19:
We are putting QM metrics into the db on the premise that in reducing very large datasets we'd want to use those values to downselect.  But this is currently only possible on site.  

I'm thinking we could periodically dump the entire db (maybe in time with data releases) and serve it from a psql server on the regular internet.

dannyjacobs, 2017-12-18:
@pkgw if you can get it dumping into the librarian and syncing back to Socorro, then we can investigate putting it on the internet somewhere for plotting purposes.

pkgw, 2017-12-19:
The database is now being backed up and synced to NMASC. Is that sufficient to close this?

miguelfmorales, 2017-12-19:
We should discuss (whenever our next meeting is). My gut feeling is no?

I think there are two pieces to the backup issue. One is to make sure we have duplication of critical data. For HERA this oddly means only a sub-sample of the real data (the stuff we move to NRAO), and the meta-data (mostly the database). So I think we've mostly solved this part.

But the other part of the backup was about enabling us to get running quickly if the key machine(s) died. This is both about reloading key data (the database), and getting the full software stack back up and running. I'm concerned about the software stack. I don't know the shape of this, but it wouldn't surprise me if after a machine failure it would take us a ~week to get the software all back up and running. From reinstalling all the libraries, getting all the dependencies, filling in the chron jobs, getting all the network setting right, etc., etc., etc. I worry that it is a rather complex software edifice, and that we have no functional record of all the necessary settings. 

So if we were confident that we could be up and running ~8 hours after plugging in a bare metal replacement, I'd say we close it. The question is what would actually be involved in getting us back up after an outage?

bhazelton, 2017-12-19:
@miguelfmorales I think you're talking about something very different than @dannyjacobs made this issue for, so I'd advocate that you make a new issue that specifies what you're talking about. This issue was about making the M&C db available off site so that we could plot things out of it (like temperatures) without hitting the main db.

dannyjacobs, 2017-12-19:
Agree with @pkgw, this particular issue is for an off-site mirror.  However
we can't close till we have an actual mirror. Right now what we have is an
archival backup.

closed 2018-12-14 by bhazelton