HERA-Team/hera_mc #158: Discuss Engineering vs. Observing data split, and associated array control pattern.

miguelfmorales, 2017-10-19:
In the next season's system there will be a number of new capabilities that will need to be actively controlled by M&C. Two specific ones are the ability to switch from sky to temperature loads at the front end, and the ability to change the analog signal level via programmable attenuators in the post amplifier module (both via control lines from the SNAP). This opens a number of system level design issues. After conferring with a number of folks, I'd like to suggest the following control pattern and associated system upgrades:

A) No changes to the signal path are _ever_ made during a science observation.

B) Creation of 'engineering' data type, that is traced through the system. This data would be recorded as usual, but be excluded from the RTP processing and any downstream science analysis (see issue #156).

C) Be able to command the correlator to stop writing a file and start a new file. (Jack is on-board with this, all baselines would have same number of time integrations). 

D) Add comments to files via M&C.

The pattern would then be to switch into engineering mode for any changes. To change the attenuation (for example), M&C would tell the correlator to stop writing the current observing file and start an engineering file. It would then tell the correlator to tell the PAM to change the attenuation, with very lax timing requirements (~1 s), then after confirmation would tell the correlator to end the engineering file and start a new observing file. All changes would be confined to engineering files.

The pro for this is that it greatly reduces the absolute timing constraints, which are a weakness of the current correlator and front end system. If you are in engineering mode, you don't really care when the thermal load was switched in/out or the attenuation changed. sub-second accuracy and concurrency of changes across the array is fine (need to make sure this is true for all such use cases). 

In contrast, if state changes are during normal observing, then we need to know when they happened to high accuracy so they can be corrected for or filtered out, ideally at sample level accuracy (2 pico-seconds). While this level of timing is doable, the correlator's timing system was not designed with this in mind. Particularly as all of our commands to the PAMs and front ends must go through the correlator, this would require a major re-engineering of the correlator timing system. The proposed system is a way to hide the command latency within non-science data taking sessions, and work around the absolute timing weakness of the current correlator. 

The last piece is a way of commenting on data, particularly engineering data, for future use. For example, tagging a data set as 'temperature load testing with load pattern alpha' vs. 'changing the attenuation' allows easier filtering to find the data to run through semi-automated processing (say a jupyter notebook).

bhazelton, 2018-12-14:
A) is taken care of in the correlator interface -- commands to change these setting can't happen while datat is being taken.

B) we now have a `tag` for correlator data taking that can be set to 'science' or 'engineering'. This tag is recorded in M&C and is supposed to be written in the data files as a header element. This part isn't happening yet.

C) this exists.

D) we don't know how/what to implement for this yet.

bhazelton, 2019-09-09:
Closing for now as A, B & C are done. D can be done after the fact if it turns out to be desired.

closed 2019-09-09 by bhazelton

dannyjacobs, 2019-09-10:
Fwiw I’ve got a prototype observing comment system to sell the group. Might
account for some of D if we do it right.