HERA-Team/hera_mc #174: Somehow get ability to tell if HERA container is drawing power

bhazelton, 2017-11-27:
Added at @dannyjacobs request.

I'm not sure what the path here is.

bhazelton, 2017-11-27:
The value is that if we can't talk to anything over the network this would let us know that the container is on and drawing power.

dannyjacobs, 2017-11-28:
How about an "on" light for the top of the container?  This would work at night anyway. @KathrynRosie if we put a bright blinky light, could we get access to the camera on the top of losberg?

dannyjacobs, 2017-12-12:
@david-deboer maybe @jack-h can take something. This would be hugely helpful.

jack-h, 2018-09-08:
There are now multiple independent switches in the container, making it very unlikely that the difference between a network failure and an electrical failure can't be determined. Can we close this?

dannyjacobs, 2018-09-09:
Any suggestions?  So far best idea is a bright light mounted on the container which would be visible from the camera on the mountain.

bhazelton, 2018-09-21:
On the Software telecon we decided to close this. @dannyjacobs your comment above seemed like a non-sequitur following Jack's comment. If you want to re-open this, please elaborate and address Jack's point.

closed 2018-09-21 by bhazelton