HERA-Team/hera_mc #186: need a simple way to plot data in M&C

dannyjacobs, 2017-12-12:
weather, uptime, whatever
We'd like to plot things without having to dump to  file and bring to matlab

dannyjacobs, 2017-12-12:
cron job dump psql db -> upload to a server somewhere -> another instance of psql -> things can connect

Items:
 - automatic backup?  This is already an open issue. Maybe use the librarian. @pkgw 
 - librarian automatic move to nrao @pkgw 
 - process at NRAO to put in another psql instance somewhere
 - plotter code using hera_mc and plotly
 - running docker instance with standard pythonstack @nkern

dannyjacobs, 2019-01-17:
Adam and I are having early success experimenting with [grafana](grafana.net).   

Current issues are are:
1) need time in linux or sql datetime, most elegant solution (and one initially proposed back during M&C design) was to make a view or meta column giving different units. Making this an M&C issue
2) some way to connect from cloud to site db. either indirectly through some streaming backup type thing, or directly with a port

Test:
https://snapshot.raintank.io/dashboard/snapshot/5J5GUHEM7kGFjNVcZI8wqekBzciWphsL&sa=D&source=hangouts&ust=1547823590508000&usg=AFQjCNFM_AWN_F0bhIvIprVuvBv6axd70Q

bhazelton, 2019-09-09:
@dannyjacobs should this still be open or is Grafana the answer?

dannyjacobs, 2019-09-09:
Grafana + simple-dashboard scripts is the answer.

closed 2019-09-10 by bhazelton