HERA-Team/hera_mc #210: sqlite db too big for Github.

david-deboer, 2018-02-14:
The sqlite version is too big for GitHub, so I need to remove most of the tables and just leave cm. Anything else? Here are the tables:

                    List of relations
 Schema |            Name             |   Type   | Owner 
--------+-----------------------------+----------+-------
 public | alembic_version             | table    | hera
 public | ant_metrics                 | table    | hera
 public | array_metrics               | table    | hera
 public | autocorrelations            | table    | hera
 public | autocorrelations_id_seq     | sequence | hera
 public | cm_version                  | table    | hera
 public | connections                 | table    | hera
 public | dubitable                   | table    | hera
 public | dubitable_start_gpstime_seq | sequence | hera
 public | geo_location                | table    | hera
 public | hera_obs                    | table    | hera
 public | lib_files                   | table    | hera
 public | lib_raid_errors             | table    | hera
 public | lib_raid_errors_id_seq      | sequence | hera
 public | lib_raid_status             | table    | hera
 public | lib_remote_status           | table    | hera
 public | lib_server_status           | table    | hera
 public | lib_status                  | table    | hera
 public | metric_list                 | table    | hera
 public | paper_temperatures          | table    | hera
 public | part_info                   | table    | hera
 public | parts                       | table    | hera
 public | roach_temperature           | table    | hera
 public | rtp_process_event           | table    | hera
 public | rtp_process_record          | table    | hera
 public | rtp_server_status           | table    | hera
 public | rtp_status                  | table    | hera
 public | rtp_task_resource_record    | table    | hera
 public | station_type                | table    | hera
 public | subsystem_error             | table    | hera
 public | subsystem_error_id_seq      | sequence | hera
 public | weather_data                | table    | hera

david-deboer, 2018-02-14:
Can people add the tables they might wish to have locally? (But no big ones, so no e.g. weather). I'll start with cm:

- cm_version
- connections
- dubitable
- geo_location
- part_info
- parts
- station_type

bhazelton, 2018-02-14:
I'd vote for just the cm tables. I think that's all the SQL lite DB was intended for, so that people could get antenna locations and connection information out of the DB easily. Anything that has rows added on a regular basis (obs, weather, etc) should be left out.

plaplant, 2018-02-17:
I agree with @bhazelton. I also think having the dubitable list is good. RTP queries it, so we'd want it to be available at NRAO for reanalysis.

bhazelton, 2018-02-21:
I think this is done, right @david-deboer?

david-deboer, 2018-02-21:
Done

closed 2018-02-21 by david-deboer