HERA-Team/hera_mc #226: connection_dossier and parts_dossier

david-deboer, 2018-05-25:
Make these confusing dictionaries into classes.

pkgw, 2018-08-08:
Noticed while testing the Python 3 changes: there are still a few places that reference the function `get_connection_dossier`, which has been removed.

```
hera_mc $ git grep get_connection_dossier |grep -v Deprecated
hera_mc/part_connect.py:        CD = h.get_connection_dossier([conn[0]], conn[1], conn[2], at_date, True)
scripts/add_connection.py:    up_check = handling.get_connection_dossier(hpn_list=[args.uppart], rev=args.uprev,
scripts/add_connection.py:    dn_check = handling.get_connection_dossier(hpn_list=[args.dnpart], rev=args.dnrev,
```

When renaming or removing internal functions it helps a *lot* to use `git grep` to make sure you got all instances.

david-deboer, 2018-08-13:
Recent merge fixed.

closed 2018-08-13 by david-deboer