HERA-Team/hera_mc #249: python3 Unicode error in pickle calls

david-deboer, 2018-09-03:
I don't know how widely this is in there, but here is one of those encoding errors:

In [1]: from hera_mc import sys_handling

In [2]: h=sys_handling.Handling()

In [3]: h.get_cminfo_correlator()
---------------------------------------------------------------------------
UnicodeDecodeError                        Traceback (most recent call last)
~/miniconda3/lib/python3.6/site-packages/numpy/lib/format.py in read_array(fp, allow_pickle, pickle_kwargs)
    656         try:
--> 657             array = pickle.load(fp, **pickle_kwargs)
    658         except UnicodeError as err:

UnicodeDecodeError: 'ascii' codec can't decode byte 0x80 in position 3: ordinal not in range(128)

During handling of the above exception, another exception occurred:

UnicodeError                              Traceback (most recent call last)
<ipython-input-3-e78d4dc987a8> in <module>()
----> 1 h.get_cminfo_correlator()

~/miniconda3/lib/python3.6/site-packages/hera_mc/sys_handling.py in get_cminfo_correlator(self)
    250         cofa_loc = self.geo.cofa()[0]
    251         stations_conn = self.get_all_fully_connected_at_date(
--> 252             at_date='now', station_types_to_check=cm_utils.default_station_prefixes)
    253         stn_arrays = StationInfo('init_arrays')
    254         for stn in stations_conn:

~/miniconda3/lib/python3.6/site-packages/hera_mc/sys_handling.py in get_all_fully_connected_at_date(self, at_date, station_types_to_check)
    147         for st in station_types_to_check:
    148             for stn in self.geo.station_types[st]['Stations']:
--> 149                 station_info = self.get_fully_connected_location_at_date(stn=stn, at_date=at_date)
    150                 if station_info is not None:
    151                     station_conn.append(station_info)

~/miniconda3/lib/python3.6/site-packages/hera_mc/sys_handling.py in get_fully_connected_location_at_date(self, stn, at_date)
    181         else:
    182             H = cm_hookup.Hookup(at_date, self.session)
--> 183         hud = H.get_hookup(hpn_list=[stn], exact_match=True)
    184         station_info = None
    185         fully_connected = cm_revisions.get_full_revision(stn, hud)

~/miniconda3/lib/python3.6/site-packages/hera_mc/cm_hookup.py in get_hookup(self, hpn_list, rev, port_query, exact_match, levels, force_new, force_specific, force_specific_at_date)
    341         # Check/get the appropriate hookup dict
    342         # (a) in memory, (b) re-read cache file, or (c) generate new
--> 343         self.determine_hookup_cache_to_use(force_new=force_new)
    344 
    345         # Now build up the returned hookup_dict

~/miniconda3/lib/python3.6/site-packages/hera_mc/cm_hookup.py in determine_hookup_cache_to_use(self, force_new)
    279 
    280         """
--> 281         cache_file_date_OK = self.hookup_cache_file_date_OK()
    282         if force_new or not cache_file_date_OK:
    283             self.cached_hookup_dict = self.get_hookup_from_db(hpn_list=self.hookup_list_to_cache, rev='ACTIVE',

~/miniconda3/lib/python3.6/site-packages/hera_mc/cm_hookup.py in hookup_cache_file_date_OK(self, contemporaneous_minutes)
    555         if os.path.exists(self.hookup_cache_file):
    556             with open(self.hookup_cache_file, 'rb') as f:
--> 557                 cached_at_date = Time(np.load(f).item())
    558         else:
    559             return False

~/miniconda3/lib/python3.6/site-packages/numpy/lib/npyio.py in load(file, mmap_mode, allow_pickle, fix_imports, encoding)
    431             else:
    432                 return format.read_array(fid, allow_pickle=allow_pickle,
--> 433                                          pickle_kwargs=pickle_kwargs)
    434         else:
    435             # Try a pickle

~/miniconda3/lib/python3.6/site-packages/numpy/lib/format.py in read_array(fp, allow_pickle, pickle_kwargs)
    661                 raise UnicodeError("Unpickling a python object failed: %r\n"
    662                                    "You may need to pass the encoding= option "
--> 663                                    "to numpy.load" % (err,))
    664             raise
    665     else:

UnicodeError: Unpickling a python object failed: UnicodeDecodeError('ascii', b'\x00\x00\x00\x80~\xc1BA', 3, 4, 'ordinal not in range(128)')
You may need to pass the encoding= option to numpy.load

pkgw, 2018-09-03:
Hmm, from my read of the code, it looks like the current incarnation should be reliable. Does this problem continue to happen if you blow away your hookup cache file?

david-deboer, 2018-09-04:
Fixed by writing two versions depending on the python version.

closed 2018-09-04 by david-deboer