HERA-Team/hera_mc #272: interfaces needed for commissioning

dannyjacobs, 2018-10-23:
plot of adc power vs dish location labeled FEM numbers
plot of adc power vs PAM 
plot of adc power vs snap slot
print out of cm connection table with adc power 

In the past we talked directly to redis for the power and M&C for the configuration.  This time we should talk directly to M&C for everything, but it means we need to get all those paths in place.

Some examples:
[image]
[image]

bhazelton, 2019-09-09:
@dannyjacobs should this still be open?

dannyjacobs, 2019-09-09:
We have more specific issues tracking this, ok to close.

closed 2019-09-10 by bhazelton