HERA-Team/hera_mc #281: geo.py plots current antennas regardless of date

dannyjacobs, 2018-11-04:
For example the command
`geo.py -g --date=2016/11/08`
returns
[image]

dannyjacobs, 2018-11-04:
I think its line 390 in  geo_handling.plot_station_types
fig_num = self.plot_stations(self.station_types[st]['Stations'], query_date, **kwargs)

I think that which gets defined as "Stations" in the station_types is not recomputed depending on the date. Either the station_types should be recomputed or plot_stations should use the query_date is its passed to filter out things which don't yet exist.  
Ideally the filter would take place at one obvious place.

david-deboer, 2018-11-05:
I'm simplifying this (hopefully synonymous with fixing...) hopefully by tomorrow.

PR #284 (2018-11-07): Iss 281
This implements the fix and makes a number of related clean-up changes.

fixes #281
closed 2018-11-07 by dannyjacobs