HERA-Team/hera_mc #286: part listing for H0C

dannyjacobs, 2018-11-14:
Trying to print out a list of which antennas had FEM or NRAO Baluns (this to keep track during our Tsys analyses).  I tried the following
`scripts/hookup.py --date=2016/3/10 `  
and got the following error

Traceback (most recent call last):
  File "scripts/hookup.py", line 59, in <module>
    force_specific_at_date=date_query)
  File "/Users/djacobs/anaconda/envs/HERA/lib/python2.7/site-packages/hera_mc-0.1.dev0-py2.7.egg/hera_mc/cm_hookup.py", line 334, in get_hookup
    requested_list_OK_for_cache = self.double_check_request_for_cache_keys(hpn_list)
  File "/Users/djacobs/anaconda/envs/HERA/lib/python2.7/site-packages/hera_mc-0.1.dev0-py2.7.egg/hera_mc/cm_hookup.py", line 373, in double_check_request_for_cache_keys
    for x in hpn_list:
TypeError: 'NoneType' object is not iterable

Am I doing it wrong, or is the code?

david-deboer, 2018-11-14:
You need to give it a list of parts:  `hookup.py -p HH --date 2016/3/10`

dannyjacobs, 2018-11-14:
Aha. According to the help -p is a required parameter so should probably
Error if a -p is not given.

david-deboer, 2018-11-14:
I've changed it to use the default list HH,HA,HB

david-deboer, 2018-11-14:
Closed with merge.

closed 2018-11-14 by david-deboer