HERA-Team/hera_mc #306: need time in linux seconds or sql datetime as a column (for grafana)

dannyjacobs, 2019-01-17:
per issue #186 (getting an online dashboard of M&C) we need tables of interest to have time in linux seconds or sql datetime. Currently we have tested this by making a view table with linux time (which is easy to derive from gps seconds by adding 315964782 (number of seconds endured by those who lived through the 70s).  Ideally this would automatically happen for all time columns.

bhazelton, 2019-01-18:
This is probably the best example to look at: 
https://bitbucket.org/zzzeek/sqlalchemy/wiki/UsageRecipes/Views

bhazelton, 2019-01-22:
Ok, I have a first example in PR #312. @dannyjacobs can you check it out and see if it's what you want?

bhazelton, 2019-02-25:
On the back burner. Apparently there's a way to take care of this issue within Grafana, but Danny's not totally happy with Graphana. Will re-evaluate what's needed once we settle on the solution.

bhazelton, 2019-09-09:
@dannyjacobs are you satisfied with the solution you have? Should this still be open?

dannyjacobs, 2019-09-09:
I am satisfied.  If grafana falls apart, we'll make a new plan.

closed 2019-09-10 by bhazelton