HERA-Team/hera_mc #307: correlator monitor interface

bhazelton, 2019-01-18:
bit statistics, etc

bhazelton, 2019-02-11:
First priority: SNAP power

bhazelton, 2019-03-05:
SNAP gain levels

jack-h, 2019-03-11:
Everything available in hera.today is now available as of https://github.com/HERA-Team/hera_corr_cm/commit/2e8ed0d03240b1c9db82545f799c9a353673f8d0

This include:
SNAP:
 - temperatures
 - uptime stats
 - PSU warning flags

per-antenna:
 - ADC mean
 - ADC power
 - ADC rms
 - digital EQ coefficients
 - PAM power levels
 - PAM attenuator settings
 - ADC histograms

FEM stats (and further PAM stats) to come in the next two weeks after liaising with @ianmalcolm

bhazelton, 2019-03-26:
ok, with PRs #331 & #333 these should be getting into the M&C database (@dannyjacobs )

bhazelton, 2019-04-08:
DDB asks: Is there another analog gain before the digitizer? @jack-h

bhazelton, 2019-04-08:
I don't think we actually have the ADC histograms, do we?

bhazelton, 2019-05-06:
What we are still missing:
- ADC histograms
- correlator input stats
- FEM sensors (temp, acceleration, ...)
- PAM temperatures

bhazelton, 2019-07-22:
with PR #370, we have ADC histograms, some more FEM & PAM sensors. 

Still missing:
- SNAP CPU uptime, IP and Mac addresses
- acceleration
- tilt
- barometer
- PAM temperatures
- X engine stats: timing, packet stats...
- correlator bit statistics: not sure if there's anything else here, maybe overflows?

bhazelton, 2019-09-09:
@jack-h are any of these easy to make available?

bhazelton, 2020-01-08:
tilt was addressed in #427 

I'm unsure whether any of the rest of these are real things we need.