HERA-Team/hera_mc #325: SNAP gain command interface

bhazelton, 2019-03-05:
commands sent to the SNAPs to adjust gain

bhazelton, 2019-04-08:
we are already capturing some related statuses (in antennna_status table):
 - pam_atten
 - eq_coeffs (digital, per frequency)

jack-h, 2019-05-08:
The correlator CM class now has access to it's own correlator instance (the thing I use in the behind-the-scenes control scripts) -- https://github.com/HERA-Team/hera_corr_cm/blob/dcee7e3a9f0d2e1edf8165a44e7f6dd5872aa103/hera_corr_cm/hera_corr_cm.py#L60

This can be used to set EQ / pam attenuation via the `set_pam_attenuation` and `set_eq` methods. https://github.com/HERA-Team/hera_corr_f/blob/a6c1a55a093e4d788408d71e6265e39a629860a7/control_software/src/hera_corr.py#L349

In fact, if you dig into this object you'll find you can pretty much do anything, but I would suggest avoiding methods on the new corr object unless you are confident they won't break something. Methods on this object generally interact directly with the boards (my intention was to use redis to prevent ever needing to do this, but I think given how much stuff is going to end up getting exposed it's probably easier to abandon that plan).

bhazelton, 2019-07-22:
I think we may need a different solution in light of the redis connection problems we had when this object was loaded by default.

bhazelton, 2019-09-09:
@jack-h any updates on this?

jack-h, 2019-10-02:
Added in https://github.com/HERA-Team/hera_corr_cm/commit/7607c1518ac11b55dcc53c6a12283a212a431bda

Note that this hasn't been installed on site

bhazelton, 2019-10-02:
Thank you @jack-h ! Some questions:
1) Are these the same equalizations that are reported via `get_ant_status`?
2) How have these been set in the past?

jack-h, 2019-10-02:
1. Yes (I'm working on adding the PAM attenuation values as well)
2. As part of the `start` call, an equalization is attempted. I'd like to turn this off and then the correlator will just use the last known values unless you explicitly change them with the new calls

jack-h, 2019-10-02:
And, pam attenuator get/set methods as of https://github.com/HERA-Team/hera_corr_cm/commit/a19a971b5f08e3c18b9e5c843ec9f7cc1becd039

dannyjacobs, 2019-10-03:
> And, pam attenuator get/set methods as of HERA-Team/hera_corr_cm@a19a971

Equalization is dominated by antennas dropping in and out, which seems to
change day to day.
Ideally you'd only auto-equalize antennas that are connected. Once you've
found the level for a working signal chain it should stay the same.

-- 
================================================================
Daniel C. Jacobs
KE7DHQ
Assistant Professor
Arizona State University
School of Earth and Space Exploration
Low Frequency Cosmology
Phone:           (505) 500 4521
Homepage:   danielcjacobs.com
MWA:   mwatelescope.org
HERA:   reionization.org
PAPER: eor.berkeley.edu