HERA-Team/hera_mc #367: time averaging of FEM tilt data

dannyjacobs, 2019-07-17:
Scott Dynes (MIT) requests being able to time-average tilt data before recording it to the database.

bhazelton, 2019-07-17:
This needs to happen in the correlator @jack-h

sdynes, 2019-07-17:
Grafana allows the time averaging of data via a detail setting in setting up  a view (at least it does for TESS data).  Not sure how flexible this is - looks a lot like an SQL cmd that gets edited.

dannyjacobs, 2019-07-17:
Yes that is correct, but the current sampling rate output by the correlator
is ~60s. So that will work as you don't mind averaging into ~30 minute
bins.  What sorts of timescales are we after here?  Wind gusts happen
fairly sharply in time, but might be source of the largest deviations.  On
the other hand we could be looking for the change in average tilt before
and after some event, in which case averaging over an hour might be ok.

bhazelton, 2019-07-25:
I should note that M&C currently doesn't have access to the tilt or barometer data from the FEMs, they haven't been made available via hera_corr_cm.

bhazelton, 2019-10-02:
We failed to have a conversation about this at HAM. The problem is that all the sensors are read out in one large loop that starts again as soon as the previous iteration finishes. Currently that results in  measurements of each sensor every ~30s. There are 2 options:
1) add a reading of the accelerometer sensors in the loop. This may slightly increase the loop time but will result in readings of the accelerometers on approximately a 30s interval.
2) add multiple readings of the accelerometers in the loop, one after another, then average those together and report the average. This would increase the loop time more significantly and would result in a slightly funny sampling function (the average of many ~1s samples every ~1 minute?).

My bias is to do the first for a couple reasons. One is that if the concern is large outlier samples, averaging outliers biases a whole set. If, on the other hand, the concern is about noise and all we care about is the DC term, sampling at a 30s cadence should be adequate. Finally, I don't want to significantly decrease the sampling rate on the other sensors.

@sdynes I think you were one of the people advocating for time averaging this sensor data, what do you think? Is there someone else we should involve in the discussion?

nimarazavi, 2019-10-03:
@bhazelton can we have a conversation to discuss the details here.  Tian has now left, so I can hopefully help with some of this.

jack-h, 2019-10-04:
These are now at least reported as of https://github.com/HERA-Team/hera_corr_cm/commit/12bb8ca46ba809551fe032e2e1ba23bfb3e112e6

I await any instructions regarding averaging.

bhazelton, 2019-10-09:
Thanks @jack-h. @nimarazavi & @sdynes how do you feel about the first option?

sdynes, 2019-10-21:
The options:

1. add a reading of the accelerometer sensors in the loop. This may slightly increase the loop time but will result in readings of the accelerometers on approximately a 30s interval.
2. add multiple readings of the accelerometers in the loop, one after another, then average those together and report the average. This would increase the loop time more significantly and would result in a slightly funny sampling function (the average of many ~1s samples every ~1 minute?)

I think there are two numbers we want from these measurements. One is the 'DC'   tilt of the FEM (i.e. the tilt from vertical given no wind); this will have to be tied back to some (assumed) calibration of the tilt measure for each FEM. I think this data would come from long averages of tilt data taken at any interval.

The other measure would be an RMS tilt (from the wind forcing feed wobbles). Not sure how this would be used at this point, but the RMS tilt would seem to be important as a quality metric. This could potentially be estimated from videos of the feeds dancing in the wind or Alec's photogrammetry effort.

I think the DC measure would be a good place to start. Alternative 1 would support this, as long as we have a way of getting the data out to average the data.

bhazelton, 2019-10-21:
Ok, sounds good. In the absence of any objections, let's do this.

@jack-h it sounds like you might already have something like this in? So I just need to add them to the DB?

jack-h, 2019-10-21:
Correct [i think/hope]. `fem_imu_theta` and `fem_imu_phi` are polled and
made available in `HeraCorrCM.get_ant_status()`.

bhazelton, 2020-01-08:
done in #427

closed 2020-01-08 by bhazelton