HERA-Team/hera_mc #368: get_part_at_station_from_type and probably other functions take a very long time over high latency connections

dannyjacobs, 2019-07-17:
Minimal reproduction:
make a tunnel from qmaster to the US
run geo.py -g  or similar
it takes many hours.

dannyjacobs, 2019-07-17:
We think its the caching that takes so long.  My minimal test just finished, took almost exactly 3 hours.   See below.

The following test runs get_part_at_station_from_type once.
(HERA) djacobs@enterprise:~$ time python tmp/hera_mc_get_part_test.py
{'HH0:A': {'e': 'PAM022', 'n': 'PAM022'}}                                                                                                                                                                                           real    178m3.404s                                                                                                user    0m53.776s
sys     0m2.800s

david-deboer, 2021-11-30:
This was fixed some time ago.

closed 2021-11-30 by david-deboer