HERA-Team/hera_mc #372: mc_monitor_correlator is erroring

dannyjacobs, 2019-07-27:
Jul 27 19:32:09 qmaster mc_monitor_correlator.py[1203]: Sat Jul 27 19:32:09 2019 -- error calling command add_antenna_status_from_corrcm
Jul 27 19:32:09 qmaster mc_monitor_correlator.py[1203]: Traceback (most recent call last):
Jul 27 19:32:09 qmaster mc_monitor_correlator.py[1203]:   File "/home/obs/anaconda/envs/HERA/bin/mc_monitor_correlator.py", line 45, in <module>
Jul 27 19:32:09 qmaster mc_monitor_correlator.py[1203]:     getattr(session, command)()
Jul 27 19:32:09 qmaster mc_monitor_correlator.py[1203]:   File "/home/obs/anaconda/envs/HERA/lib/python2.7/site-packages/hera_mc/mc_session.py", line 3223, in add_antenna_status_from_corrcm
Jul 27 19:32:09 qmaster mc_monitor_correlator.py[1203]:     ant_status_dict=ant_status_dict)
Jul 27 19:32:09 qmaster mc_monitor_correlator.py[1203]:   File "/home/obs/anaconda/envs/HERA/lib/python2.7/site-packages/hera_mc/correlator.py", line 913, in create_antenna_status
Jul 27 19:32:09 qmaster mc_monitor_correlator.py[1203]:     fem_id = _pam_fem_serial_list_to_string(ant_dict['fem_id'])
Jul 27 19:32:09 qmaster mc_monitor_correlator.py[1203]:   File "/home/obs/anaconda/envs/HERA/lib/python2.7/site-packages/hera_mc/correlator.py", line 855, in _pam_fem_serial_list_to_string
Jul 27 19:32:09 qmaster mc_monitor_correlator.py[1203]:     for int_val in serial_number_list:
Jul 27 19:32:09 qmaster mc_monitor_correlator.py[1203]: TypeError: 'int' object is not iterable

dannyjacobs, 2019-07-29:
I think Yet Another Interface Issue.  When get_ant_status doesn't have a value for something it returns the string 'None' not the type None.

In [16]: for k,v in corr_status.iteritems():
    ...:     print(k,v['pam_id'])
('107:e', [112, 46, 63, 59, 1, 0, 0, 234])
('107:n', [112, 46, 63, 59, 1, 0, 0, 234])
('25:e', 'None')

But `hera_mc.correlator.create_antenna_status` checks for proper Noneness
        if ant_dict['pam_id'] is not None:
            pam_id = _pam_fem_serial_list_to_string(ant_dict['pam_id'])

I propose the following 
        if not ant_dict['fem_id'] in (None,'None'):
            fem_id = _pam_fem_serial_list_to_string(ant_dict['fem_id'])

Testing on site now.

jack-h, 2019-07-29:
FYI -- just checked https://github.com/HERA-Team/hera_corr_cm/blob/master/hera_corr_cm/hera_corr_cm.py and string None does appear to be the "correct" (i.e. documented) behaviour on the part of that code.

bhazelton, 2019-07-30:
I believe it does check for the string 'None'. See line 894 of hera_mc.correlator. It turns the string 'None' into a real `None`.

dannyjacobs, 2019-07-30:
Nope thats not it.  Leaving this one here while I poke at node9.

dannyjacobs, 2019-07-30:
Yeah, that wasn't it.  I saw the Nones in the response from hera_corr_cm and got excited. By adding a print statement I see that somehow its picking up a "-1" where a None or a list should be.  Got distracted and haven't run it to ground.   But it would be nice to get this sorted, any remote help appreciated.

PR #375 (2019-07-31): Handle -1s and other weird values in fem/pam IDs
closes #372
closed 2019-07-31 by dannyjacobs