HERA-Team/hera_mc #376: Not getting data from node temp sensors

dannyjacobs, 2019-08-01:
Bottom, middle, and top temp sensors aren't showing up in the db.

bhazelton, 2019-08-01:
Is this for all nodes or some nodes?

bhazelton, 2019-08-01:
This is what I see currently:
```
hera_mc=> select * from node_sensor order by time desc limit 10;
    time    | node | top_sensor_temp | middle_sensor_temp | bottom_sensor_temp | humidity_sensor_temp | humidity 
------------+------+-----------------+--------------------+--------------------+----------------------+----------
 1248707674 |    9 |           31.38 |              26.13 |              16.44 |                 20.4 |     8.69
 1248707673 |    8 |           27.13 |              24.81 |              20.06 |                22.65 |    16.27
 1248707673 |    0 |                 |                    |              20.13 |                23.07 |     6.85
 1248707613 |    9 |           31.44 |              26.13 |              16.44 |                20.41 |     8.66
 1248707613 |    8 |           27.06 |              24.81 |              20.13 |                22.65 |    16.23
 1248707613 |    0 |                 |                    |              20.13 |                22.98 |      6.8
 1248707553 |    9 |           31.38 |              26.13 |              16.44 |                20.41 |     8.92
 1248707553 |    8 |              27 |              24.81 |              20.19 |                22.67 |    16.27
 1248707552 |    0 |                 |                    |              20.19 |                22.98 |     6.81
 1248707493 |    9 |           31.38 |              26.13 |              16.44 |                20.43 |      8.8
(10 rows)
```

bhazelton, 2019-08-01:
So it looks like just node0 is not reporting the top or middle sensor (but is reporting the bottom). I have a vague memory that node0 might be missing some of these, but that might be wrong. @jack-h do you have any insight?

bhazelton, 2019-08-01:
And it looks like new rows are showing up regularly...

dannyjacobs, 2019-08-01:
You are right! I was only looking at node 0 but node8 is working perfectly.  I've added all the sensors to the dashboard. Looks good!
http://enterprise.sese.asu.edu:8484/d/fXTf7MWWk/nodes?from=now-24h&to=now&refresh=10s&var-the70s=315964782&var-node=00&var-node=09

closed 2019-08-01 by bhazelton