HERA-Team/hera_mc #381: correlator monitor is erroring.

dannyjacobs, 2019-08-19:
Traceback (most recent call last):
  File "/home/obs/anaconda/envs/HERA/bin/mc_monitor_correlator.py", line 45, in <module>
    getattr(session, command)()
  File "/home/obs/anaconda/envs/HERA/lib/python2.7/site-packages/hera_mc/mc_session.py", line 3223, in add_antenna_status_from_corrcm
    ant_status_dict=ant_status_dict)
  File "/home/obs/anaconda/envs/HERA/lib/python2.7/site-packages/hera_mc/correlator.py", line 913, in create_antenna_status
    fem_voltage = ant_dict['fem_voltage']
  File "/home/obs/anaconda/envs/HERA/lib/python2.7/site-packages/hera_mc/correlator.py", line 855, in _pam_fem_serial_list_to_string
    serial_str = ''
TypeError: 'int' object is not iterable

bhazelton, 2019-08-19:
This makes no sense given the current master. Paul thinks it might be out of sync `.pyc` files in site-packages. We're going to stop the daemons, remove the `hera_mc` folder in site-packages, reinstall and restart the daemons.

bhazelton, 2019-08-19:
We did those things and don't see immediate errors in /var/log/syslog. Closing for now, please reopen if needed.

closed 2019-08-19 by bhazelton