HERA-Team/hera_mc #387: Fix/add to write_records_to_file script.

david-deboer, 2019-08-30:


david-deboer, 2019-08-30:
The incorrect entry is 'ant_metric', which should be 'ant_metrics' and also the method etc are wrong (look to be cut-pasted from entry above).

Not all of the missing tables need to be covered.  For completeness, here is a list of the missing tables, some of which should probably be included:
alembic_version
apriori_antenna
array_metrics
autocorrelations
cm_version
connections
correlator_config_file
dubitable
geo_location
metric_list
paper_temperatures
part_info
parts
roach_temperature
station_type

bhazelton, 2019-08-30:
@david-deboer most of these tables don't have an existing method on MCSession that calls the `_time_filter` method, which is required under this script's paradigm. That actually includes `ant_metrics` so it should not have been included in the dict at all.

bhazelton, 2019-08-30:
I will make a PR to take it out entirely. We could add an appropriate method, but I'll leave that until someone wants it.

david-deboer, 2019-08-31:
PR removing dict entry merged with master.

closed 2019-08-31 by david-deboer