HERA-Team/hera_mc #391: NSF milestone to record RTP/OPM hash to database

david-deboer, 2019-09-04:


bhazelton, 2019-09-06:
@plaplant a couple options here:
1) add it to the rtp_status table (other columns are: time, status, event_min_elapsed, num_processes, restart_hours_elapsed). Upside: not a separate table. Downside: the git hash probably doesn't change often, so this table would have a lot of the same value in it.
2) Make a new table, probably with a start & stop time along with a hash. When the hash changes, update the stop time to the current time, then make a new row with the new hash. Upside: can be as few rows as needed. Downside: more complicated updating process, although the logic only needs to be written once.

Thoughts?

plaplant, 2019-09-06:
I have a preference for (2), because (1) seems overkill for how infrequently the repo will end up changing. But I can also see the merit in keeping it all in the rtp_status table, so if others prefer that option I'd be okay with that.

bhazelton, 2019-09-09:
Plan is to do 2. I will provide a method that RTP can call with its current hash which will update the table if needed.

bhazelton, 2019-09-09:
We actually realized we already have the structure we need in M&C to handle this and more. The problem is that with the changeover to OPM, none of the RTP reporting tables are being filled out. This will be a fair bit of work to remedy.

bhazelton, 2019-10-30:
@plaplant is this fixed now?

david-deboer, 2020-04-27:
Matt K. verified we are doing this.

closed 2020-04-27 by david-deboer