HERA-Team/hera_mc #392: sqlite doesn't match db schema

dannyjacobs, 2019-09-05:
After updating hera_mc and hera_mc sqlite no longer matches the expected schema. Is this not tested in the CI?

```
(HERA) djacobs@hayek:hera_mc (master)$ geo.py -g --label=num
ERROR:hera_mc:Model <class 'hera_mc.autocorrelations.Autocorrelations'> declares table autocorrelations which does not exist in database Engine(sqlite:////Users/djacobs/src/hera_cm_db_updates/hera_mc.db)
ERROR:hera_mc:Model <class 'hera_mc.cm_partconnect.Parts'> declares table parts which does not exist in database Engine(sqlite:////Users/djacobs/src/hera_cm_db_updates/hera_mc.db)
ERROR:hera_mc:Model <class 'hera_mc.cm_partconnect.AprioriAntenna'> declares table apriori_antenna which does not exist in database Engine(sqlite:////Users/djacobs/src/hera_cm_db_updates/hera_mc.db)
ERROR:hera_mc:Model <class 'hera_mc.cm_partconnect.PartInfo'> declares table part_info which does not exist in database Engine(sqlite:////Users/djacobs/src/hera_cm_db_updates/hera_mc.db)
ERROR:hera_mc:Model <class 'hera_mc.cm_partconnect.Connections'> declares table connections which does not exist in database Engine(sqlite:////Users/djacobs/src/hera_cm_db_updates/hera_mc.db)
ERROR:hera_mc:Model <class 'hera_mc.geo_location.StationType'> declares table station_type which does not exist in database Engine(sqlite:////Users/djacobs/src/hera_cm_db_updates/hera_mc.db)
ERROR:hera_mc:Model <class 'hera_mc.geo_location.GeoLocation'> declares table geo_location which does not exist in database Engine(sqlite:////Users/djacobs/src/hera_cm_db_updates/hera_mc.db)
ERROR:hera_mc:Model <class 'hera_mc.cm_transfer.CMVersion'> declares table cm_version which does not exist in database Engine(sqlite:////Users/djacobs/src/hera_cm_db_updates/hera_mc.db)
ERROR:hera_mc:Model <class 'hera_mc.correlator.CorrelatorControlState'> declares table correlator_control_state which does not exist in database Engine(sqlite:////Users/djacobs/src/hera_cm_db_updates/hera_mc.db)
ERROR:hera_mc:Model <class 'hera_mc.correlator.CorrelatorConfigFile'> declares table correlator_config_file which does not exist in database Engine(sqlite:////Users/djacobs/src/hera_cm_db_updates/hera_mc.db)
ERROR:hera_mc:Model <class 'hera_mc.correlator.CorrelatorConfigStatus'> declares table correlator_config_status which does not exist in database Engine(sqlite:////Users/djacobs/src/hera_cm_db_updates/hera_mc.db)
ERROR:hera_mc:Model <class 'hera_mc.correlator.CorrelatorControlCommand'> declares table correlator_control_command which does not exist in database Engine(sqlite:////Users/djacobs/src/hera_cm_db_updates/hera_mc.db)
ERROR:hera_mc:Model <class 'hera_mc.correlator.CorrelatorTakeDataArguments'> declares table correlator_take_data_arguments which does not exist in database Engine(sqlite:////Users/djacobs/src/hera_cm_db_updates/hera_mc.db)
ERROR:hera_mc:Model <class 'hera_mc.correlator.CorrelatorConfigCommand'> declares table correlator_config_command which does not exist in database Engine(sqlite:////Users/djacobs/src/hera_cm_db_updates/hera_mc.db)
ERROR:hera_mc:Model <class 'hera_mc.correlator.CorrelatorSoftwareVersions'> declares table correlator_software_versions which does not exist in database Engine(sqlite:////Users/djacobs/src/hera_cm_db_updates/hera_mc.db)
ERROR:hera_mc:Model <class 'hera_mc.correlator.SNAPConfigVersion'> declares table snap_config_version which does not exist in database Engine(sqlite:////Users/djacobs/src/hera_cm_db_updates/hera_mc.db)
ERROR:hera_mc:Model <class 'hera_mc.correlator.SNAPStatus'> declares table snap_status which does not exist in database Engine(sqlite:////Users/djacobs/src/hera_cm_db_updates/hera_mc.db)
ERROR:hera_mc:Model <class 'hera_mc.correlator.AntennaStatus'> declares table antenna_status which does not exist in database Engine(sqlite:////Users/djacobs/src/hera_cm_db_updates/hera_mc.db)
ERROR:hera_mc:Model <class 'hera_mc.daemon_status.DaemonStatus'> declares table daemon_status which does not exist in database Engine(sqlite:////Users/djacobs/src/hera_cm_db_updates/hera_mc.db)
ERROR:hera_mc:Model <class 'hera_mc.observations.Observation'> declares table hera_obs which does not exist in database Engine(sqlite:////Users/djacobs/src/hera_cm_db_updates/hera_mc.db)
ERROR:hera_mc:Model <class 'hera_mc.subsystem_error.SubsystemError'> declares table subsystem_error which does not exist in database Engine(sqlite:////Users/djacobs/src/hera_cm_db_updates/hera_mc.db)
ERROR:hera_mc:Model <class 'hera_mc.librarian.LibServerStatus'> declares table lib_server_status which does not exist in database Engine(sqlite:////Users/djacobs/src/hera_cm_db_updates/hera_mc.db)
ERROR:hera_mc:Model <class 'hera_mc.librarian.LibStatus'> declares table lib_status which does not exist in database Engine(sqlite:////Users/djacobs/src/hera_cm_db_updates/hera_mc.db)
ERROR:hera_mc:Model <class 'hera_mc.librarian.LibRAIDStatus'> declares table lib_raid_status which does not exist in database Engine(sqlite:////Users/djacobs/src/hera_cm_db_updates/hera_mc.db)
ERROR:hera_mc:Model <class 'hera_mc.librarian.LibRAIDErrors'> declares table lib_raid_errors which does not exist in database Engine(sqlite:////Users/djacobs/src/hera_cm_db_updates/hera_mc.db)
ERROR:hera_mc:Model <class 'hera_mc.librarian.LibRemoteStatus'> declares table lib_remote_status which does not exist in database Engine(sqlite:////Users/djacobs/src/hera_cm_db_updates/hera_mc.db)
ERROR:hera_mc:Model <class 'hera_mc.librarian.LibFiles'> declares table lib_files which does not exist in database Engine(sqlite:////Users/djacobs/src/hera_cm_db_updates/hera_mc.db)
ERROR:hera_mc:Model <class 'hera_mc.node.NodeSensor'> declares table node_sensor which does not exist in database Engine(sqlite:////Users/djacobs/src/hera_cm_db_updates/hera_mc.db)
ERROR:hera_mc:Model <class 'hera_mc.node.NodePowerStatus'> declares table node_power_status which does not exist in database Engine(sqlite:////Users/djacobs/src/hera_cm_db_updates/hera_mc.db)
ERROR:hera_mc:Model <class 'hera_mc.node.NodePowerCommand'> declares table node_power_command which does not exist in database Engine(sqlite:////Users/djacobs/src/hera_cm_db_updates/hera_mc.db)
ERROR:hera_mc:Model <class 'hera_mc.rtp.RTPServerStatus'> declares table rtp_server_status which does not exist in database Engine(sqlite:////Users/djacobs/src/hera_cm_db_updates/hera_mc.db)
ERROR:hera_mc:Model <class 'hera_mc.rtp.RTPStatus'> declares table rtp_status which does not exist in database Engine(sqlite:////Users/djacobs/src/hera_cm_db_updates/hera_mc.db)
ERROR:hera_mc:Model <class 'hera_mc.rtp.RTPProcessEvent'> declares table rtp_process_event which does not exist in database Engine(sqlite:////Users/djacobs/src/hera_cm_db_updates/hera_mc.db)
ERROR:hera_mc:Model <class 'hera_mc.rtp.RTPProcessRecord'> declares table rtp_process_record which does not exist in database Engine(sqlite:////Users/djacobs/src/hera_cm_db_updates/hera_mc.db)
ERROR:hera_mc:Model <class 'hera_mc.rtp.RTPTaskResourceRecord'> declares table rtp_task_resource_record which does not exist in database Engine(sqlite:////Users/djacobs/src/hera_cm_db_updates/hera_mc.db)
ERROR:hera_mc:Model <class 'hera_mc.qm.AntMetrics'> declares table ant_metrics which does not exist in database Engine(sqlite:////Users/djacobs/src/hera_cm_db_updates/hera_mc.db)
ERROR:hera_mc:Model <class 'hera_mc.qm.ArrayMetrics'> declares table array_metrics which does not exist in database Engine(sqlite:////Users/djacobs/src/hera_cm_db_updates/hera_mc.db)
ERROR:hera_mc:Model <class 'hera_mc.qm.MetricList'> declares table metric_list which does not exist in database Engine(sqlite:////Users/djacobs/src/hera_cm_db_updates/hera_mc.db)
ERROR:hera_mc:Model <class 'hera_mc.weather.WeatherData'> declares table weather_data which does not exist in database Engine(sqlite:////Users/djacobs/src/hera_cm_db_updates/hera_mc.db)
Traceback (most recent call last):
  File "/Users/djacobs/miniconda3/envs/HERA/bin/geo.py", line 51, in <module>
    db = mc.connect_to_mc_db(args)
  File "/Users/djacobs/miniconda3/envs/HERA/lib/python3.7/site-packages/hera_mc/mc.py", line 192, in connect_to_mc_db
    db = AutomappedDB(db_url)
  File "/Users/djacobs/miniconda3/envs/HERA/lib/python3.7/site-packages/hera_mc/mc.py", line 85, in __init__
    .format(db_url))
RuntimeError: database sqlite:////Users/djacobs/src/hera_cm_db_updates/hera_mc.db does not match expected schema

```

bhazelton, 2019-09-05:
did you do an alembic upgrade?

david-deboer, 2019-09-05:
I did right now, but it didn't upgrade anything, so that is why I'm perplexed.  I went ahead and re-pushed a new one.  You need to make sure you pull down from hera_cm_db_updates.

bhazelton, 2019-09-05:
I meant, did Danny do an upgrade

dannyjacobs, 2019-09-05:
Do I need to? If so it’s not a step in install.md

david-deboer, 2019-09-05:
Operationally, one always needs to make sure that the database schema is up-to-date and hera_cm_db_updates is up-to-date?  (And if using psql that cm_init.py is run).  Do we need an 'OPERATIONS.md' file?  Typically this would go into the README file (?)

dannyjacobs, 2019-09-06:
Prior to my error: I pulled from hera_cm_db_updates, pulled and installed hera_mc.  Was there something else I was supposed to do?

dannyjacobs, 2019-09-06:
Whatever @david-deboer did to the cm_updates fixed the problem.

closed 2019-09-06 by dannyjacobs