HERA-Team/hera_mc #394: Update for new katcp portal

david-deboer, 2019-09-10:
Access for katcp for weather data is changing:

---------- Forwarded message ---------
From: Toufeeq Ockards <tockards@ska.ac.za>
Date: Thu, Sep 5, 2019 at 11:08 AM
Subject: Re: katportalclient updates [important to users of katportalclient]
To: Dave Horn <dhorn@ska.ac.za>
Cc: Lance Williams <lance@ska.ac.za>

Hi Dave,

The upgrade plan is to do a pip install katportalclient -U when we upgrade our katstore servers on site .

Currently we would like to let users do the install of katportalclient from pypi and not the githu repo. 

We will notify all users of this via email and github. 

regards,
MTO

On Thu, Sep 5, 2019 at 7:50 AM Dave Horn <dhorn@ska.ac.za> wrote:
HI Toufeeq, Lance,

Please could you clarify further the upgrade path for USE that are using katportalclient - already installed.

If they make no changes until after the new katstore is deployed to site and then update with pip install, will that work?

Will you let us know when katstore is updated on site.

Thanks

Dave

On Mon, Sep 2, 2019 at 4:08 PM Toufeeq Ockards <tockards@ska.ac.za> wrote:
Hi All,

This email is directed to anyone who uses katportalclient and installs it from the github repository advising them to change installation to pypi.org using pip install katportalclient .

Updates to the master branch of katportalclient on github are planned for this Wednesday (4 September 2019) and will affect users who do pip installs from github. As the changes will break sensor retrieval from katstore. 

pypi.org will not have the changes that are in master(until new katstore is deployed on site)  
pip install katportalclient is the newly preferred way. see https://pypi.org/project/katportalclient/ 

Please forward this email to anyone who uses katportalclient 

regards,
MTO

dannyjacobs, 2019-09-13:
Also tracked in Commissioning issue: https://github.com/HERA-Team/HERA_Commissioning/issues/198

The key API details are here: https://github.com/ska-sa/katportalclient/issues/52

dannyjacobs, 2019-09-14:
fixed in PR #395

closed 2019-10-30 by bhazelton