HERA-Team/hera_mc #448: Redefine apriori status values

mkolopanis, 2019-12-11:
A discussion at SALF brought up the idea we may need to redefine the known values for the apriori antenna status table. I think the decision was a _new_ discussion needs to happen to find a way forward for this table.

@bhazelton @nimarazavi @dannyjacobs

dannyjacobs, 2019-12-17:
Categories agreed at 17 Dec 2019 Commissioning meeting. Status sets owner.  The rule: only current dish owner allowed to change the status.
- dish_maintenance
- dish_ok
- RF_maintenance
- RF_ok 
- digital_maintenance
- digital_ok

bhazelton, 2019-12-18:
I'm sorry I couldn't make the 17 Dec 2019 Commissioning meeting, but this set doesn't make sense to me. Can we talk about it more at another meeting?

mkolopanis, 2019-12-18:
also just found this issue too about this HERA-Team/HERA_Commissioning#365

dannyjacobs, 2019-12-19:
Yeah I didn't record much for motivation, mostly because still trying to
compactly articulate things. The intention is to provide a simple
delineation of  status according to subsystem team, provide each stage with
a way to indicate "in progress" and "ready for next stage".  My goal here
is to make an incremental change to the system (ie relatively fast and easy
to integrate) that allows antennas to be "handed around".  At the same time
you need a basic rule for how that handoff happens.  The basic rule is, of
course, not enough, further documented conversation is also necessary.

The proposed rule is that only the current owner of the maintenance mode
gets to change the mode to something else at which time the antenna belongs
to the new team.  The handover has to be a two-stage process, so hence the
existence of the OK stage which indicates "ready for handoff".

bhazelton, 2020-01-10:
Add the following:

- calibration maintenance
- calibration ok
- calibration triage

david-deboer, 2020-01-10:
Two comments:
1 - do we want the underscores, or use spaces?
2 - I don't think that we want to keep the old values as valid options.  The old values can stay in the database, but going forward we only want new values recognized.

mkolopanis, 2020-01-10:
with the current implementation of the PR a nice warning is issued if a user tries to input an old value. Can leave them as acceptable field values, but not allow users to set to them if you'd like.

david-deboer, 2020-01-10:
I think that would be better.  Thanks.

sdynes, 2020-01-11:
I think the old values should not be choices moving forward  - it's a paradigm in the past.

Also: underscores.

dannyjacobs, 2020-01-14:
This is being addressed in PR #454

PR #454 (2020-01-25): Adds new apriori Statuses
Adds statuses and closes #448 
Keeps old arpiori status values for backwards compatibility but sends a DeprecationWarning if one is set.
closed 2020-01-25 by david-deboer