HERA-Team/hera_mc #460: Redis dependent code should be in one place

mkolopanis, 2020-01-28:
Brought up on 27-Jan correlator/software call. 

All code which puts things into redis should ideally be in one place whether a corr_cm/f or in hera_mc. But currently redis code in hera_corr_f needs hera_mc to put things into the redis database.
This will become an issue as hera_mc moves away from python 2 but corr_f and corr_cm continue to be python 2 only.

https://github.com/HERA-Team/hera_corr_f/blob/8d5a2fff384f74ed0c41f25de667bc3a5af606fd/control_software/src/helpers.py#L128-L174

bhazelton, 2020-01-29:
I think the question is whether this should go in hera_corr_cm or in hera_mc directly. One question I have is how should writing the info to redis be triggered? If it needs to be triggered by hera_corr_f, then the code hera_corr_f calls needs to work in python 2, which pushes towards putting this code in hera_corr_cm. If it is going to be triggered by a cron job or a daemon, it might be better to put it into hera_mc.

bhazelton, 2020-02-04:
This needs to be written in hera_mc. It will be called by a cron job or daemon ~once per day.
Need a companion documentation which is the interface control document.

bhazelton, 2020-09-15:
I believe this is done. @mkolopanis ?

mkolopanis, 2020-09-15:
I think so. we have the `redis_cm_corr` module now that writes to redis with all the cminfo.

closed 2020-09-15 by bhazelton