HERA-Team/hera_mc #465: Accessing a priori antenna table

jsdillon, 2020-02-04:
Danny suggested I file an issue here:

For redundant calibration, I need to access the apriori antenna status table to determine which antennas to exclude from calibration. This needs to be able to access the antenna status table for our best current knowledge of what the statuses were at a specific time in the past as well. I'd like to be able to do this both onsite as part of RTP and at NRAO as part of offline analysis, so it needs to be makeflow-able.

I'm unfamiliar with hera_mc. What's the proper procedure for this?

david-deboer, 2020-02-04:
In talking with Josh, it sounds like we want an up-to-date version of cm stuff at NRAO, which shouldn't be difficult via the hera_cm_db_updates route.

jsdillon, 2020-02-04:
(I'd also be fine with some sort of standardized tunneling scheme from NRAO to site.)

dannyjacobs, 2020-02-04:
Its not just cm stuff. He wants the apriori status from antenna_status.
We do have a tunnel to site, should possible to connect to that from NRAO.  It only works if the link is up, which is not having a great month.

bhazelton, 2020-02-04:
the apriori status table is part of cm

bhazelton, 2020-02-04:
I believe that `hera_cm_db_updates` has everything he needs.

david-deboer, 2020-02-04:
At the very least, for developing and testing etc we could install hera_mc at NRAO (if it isn't?) then decide how we wanted to do it operationally (I tend to avoid tunnels if there is another viable option...).

david-deboer, 2020-02-05:
Josh has the sqlite version working on his Mac (happily, it installed and worked easily;-).  I think the simplest approach is to spin up the Postgres db at NRAO and then cron update it via hera_cm_db_updates.

bhazelton, 2020-02-05:
sounds reasonable

jsdillon, 2020-02-05:
Perhaps this is a question for today's general telecon, but who is responsible for getting this set up and maintaining this cron job as heramgr on NRAO?

bhazelton, 2020-09-15:
@plaplant any progress on getting the M&C database regularly updated at NRAO?

bhazelton, 2021-04-06:
this was done awhile ago.

closed 2021-04-06 by bhazelton