HERA-Team/hera_mc #467: Write subsystem errors to/from redis

david-deboer, 2020-02-05:
Re:  hera_corr_f de-hera_mc’ing - I’ve updated hera_corr_f and will try and work it through to installation…  There is another location that calls hera_mc
class HeraMCHandler(logging.Handler):
    def __init__(self, subsystem, *args, **kwargs):
        from hera_mc import mc
        from astropy.time import Time
        self.Time = Time
        logging.Handler.__init__(self, *args, **kwargs)
        self.subsystem = subsystem
        db = mc.connect_to_mc_db(None)
        self.session = db.sessionmaker()
    def emit(self, record):
        # Re-code level because HeraMC logs 1 as most severe, and python logging
        # calls critical:50, debug:10
        severity = max(1, 100 / record.levelno)
        message = self.format(record)
        logtime = self.Time(time.time(), format="unix")
        self.session.add_subsystem_error(logtime, self.subsystem, severity, message)
I’d like to get rid of that and define redis keys and task someone to handle it on the hera_mc side (@matthew.kolopanis?)

mkolopanis, 2020-06-02:
I think we actually have resolved this now. We use a redis pub/sub to send errors into M&C database.

bhazelton, 2021-04-06:
this has been fixed some time back

closed 2021-04-06 by bhazelton