HERA-Team/hera_mc #491: Add monitoring for the deployed version for all site repos

bhazelton, 2020-06-24:
Particularly the correlator, but also all other repos installed on site. Probably want columns for machine name, environment name, repository name along with the standard git info (hash, branch, upstream etc).

dannyjacobs, 2020-08-25:
We really need this. Currently running hera_corr_f on a hacky branch to shim up the hacked fengine.

bhazelton, 2020-08-25:
Ok, so does this table schema seem right? The starred columns are part of the primary key. I guess the question is: do we ever have 2 copies of a repo on a host that we want to use? If so, the path needs to be part of the primary key.

*host
*repo
*starttime
stoptime
path
branch
hash

@dannyjacobs @plaplant @mkolopanis @david-deboer

mkolopanis, 2020-08-25:
What about environment as a (primary?) key value. It would be a string
point to the base level of the environment. Could easily track conda,
virtual env, even system level that wau.

mkolopanis, 2020-08-25:
Or is that path?

bhazelton, 2020-08-27:
*host
*repo
*starttime
*environment
stoptime
path
upstream
branch
hash