HERA-Team/hera_mc #498: Put snap hostname in M&C

dannyjacobs, 2020-08-18:
At the present time one must construct the snap hostname by looking at the node and slot locations of the snap.   There are several reasons why this is less than ideal. First it is an implicit algorithm, there is no guarantee that the correlator internally will adhere to this algorithm. ie if someone moves a snap from one node to another, they must change both M&C _and_ the hostname in hera-snap-head:/etc/hosts .  Second the query is slow and one of the reasons why the [grafana antenna ownership table](https://enterprise.sese.asu.edu:8484/d/mV2iZUPWk/apriori-status?orgId=1&refresh=1m) takes a while to load.

@bhazelton would like to find a better way.

david-deboer, 2020-08-18:
The /etc/hosts stuff above is part of the "config" stuff we spoke about.  We could contain it all in this issue.

david-deboer, 2020-08-18:
This is related to issue #499

david-deboer, 2020-08-20:
This is being done in PR #500

closed 2020-09-15 by bhazelton