HERA-Team/hera_mc #523: Incomplete redis setup for testing with empty redis db.

david-deboer, 2020-11-18:
There seems to be some redis keys that aren't set in an empty redis database for complete testing.
1 - 'hera':  this should be set up manually since it enables the local testing and can be set to any value e.g. r.set('hera', 'test')
2 - 'auto:timestamp':  something like r.set('auto:timestamp', np.asarray(2459172., dtype=np.float64).tobytes())
3 - I think there are others, because when I put in an 'auto:timestamp' that gets accepted, it still fails testing.

mkolopanis, 2020-11-18:
The tests that rely on redis assume you have the `dump.rdb` file from the repo loaded into redis as the database. I 100% agree this is not optimal though.

david-deboer, 2020-11-18:
Thanks - I do see a note at the bottom of the README.md.  I'll close the issue.

david-deboer, 2020-11-18:
To archive this fact: the snap configuration yaml is defined in tests __init__.py and set in pytests, not in the dump.rdb.

closed 2020-11-18 by david-deboer