HERA-Team/hera_mc #525: make a new table for tracking slurm job ids

bhazelton, 2020-12-15:
*obsid
*task_name
*starttime (integer second)
slurm_job_id

bhazelton, 2021-02-03:
Comparing this proposed structure to the rtp_task_resource_record table, which is the one we're trying to fill out (this one is essentially a helper for that one to track the job id), I see that this one includes the start_time in the primary key while the rtp_task_resource_record table does not. I don't remember the details on why we wanted to include that in the primary key, maybe it was about multiple runs or re-runs? But if we need it for this table do we also need it for the rtp_task_resource_record table? Or should that one just have the final run data in it? @plaplant maybe you're the best person to comment.

plaplant, 2021-02-04:
Yes, we wanted to have the start time to capture multiple potential runs of the same task. I think for the `rtp_task_resource_record`, I have a preference for keeping only the final (i.e., successful) run, but if it's easier to include all of them then let's go with that.

bhazelton, 2021-02-04:
It's not easier, what we have only allows for one record per obsidian/task and I'm fine with leaving it that way, I just wanted to check that that was the right structure.

bhazelton, 2021-03-23:
done in #529

closed 2021-03-23 by bhazelton