HERA-Team/hera_mc #527: make a table to track RTP starting

bhazelton, 2020-12-15:
*obsid
* submitted_time (GPS second integer)
jd (integer)
obs_tag ("engineering"/"science")
submitted (boolean)

We'll write a daemon that grabs all non-submitted jobs, breaks into groups of integer JDs and launches RTP on each group. RTP needs to update submitted time or make a new row if it already exists (for a re-run).

plaplant, 2021-02-26:
@bhazelton thinking about this a bit more, is this a table that will be added to as an observation is added to M&C? If so, then the `submitted_time` column might not initially be filled out, so it probably shouldn't be part of the primary key. But if it's only updated when a job is submitted, then I'm not sure the `submitted` column would be necessary (because it wouldn't be in the table if it hadn't been submitted). I'm leaning toward just removing the `submitted` column, but wanted to get your thoughts.

(Also I'm happy to take over this issue, I have a vested interest in making RTP start-up more reliable and automated 🙂 )

bhazelton, 2021-02-26:
Yep, I see the problem. One problem is that if we only have obsid as the primary key, then this table can't track re-runs. Another option would be to add a column which tracks iterations, so if it's added by an automated process when the obs is made that could be set to zero and then set to a larger integer for each re-run. Then the submitted_time could be allowed to be null until the job is submitted. It depends on whether you want this table to support re-runs or not.

PR #538 (2021-03-18): Add table to track observations run through RTP
This PR adds a new table (`RTPLaunchRecord`) which keeps track of observations that have been run through RTP. So far the PR defines the new table, and methods to add records and update them. ~Still needed are scripts to do this from the command line, as well as a related cron job to automatically launch jobs for recent observations.~ Scripts for adding/updating entries and launching RTP have been added in a834d77 and 0a57a8e.

There are also some incidental changes around [deprecated numpy datatypes](https://numpy.org/doc/stable/release/1.20.0-n [... cut]
closed 2021-03-18 by bhazelton