HERA-Team/hera_mc #615: Add armed/configured snaps to mc from snap_start.log

david-deboer, 2022-08-02:
The snap_start.log is scraped daily after hera_feng init is run and placed into redis using:
        stat = {'timestamp': datetime.datetime.now().strftime('%Y-%m-%dT%H:%M:%S'),
                'log_time_start': self.log_time_start.strftime('%Y-%m-%dT%H:%M:%S.%f'),
                'log_time_stop': self.log_time_stop.strftime('%Y-%m-%dT%H:%M:%S.%f'),
                'working': ','.join(self.working),
                'unconfig': ','.join(self.unconfig),
                'maxout': ','.join(self.maxout)}
        r.hmset('snap_log', stat)

Fields are:
   timestamp - when the data are written to redis
   log_time_start/stop - the scraper takes the latest arm entries in the log, bracketed by these times (so this is when it happened)
   working - csv list of the snap hostnames that hera_corr_f thinks work
   unconfig - ones that made it to arming, but weren't configured (so don't work)
   maxout - ones that made it all the way, but were ignored because we had too many snaps

bhazelton, 2022-08-02:
Thanks. Do we need all there time stamps in the M&C DB? I'd lean towards just the log start/stop because I'm not sure that the redis timestamp adds much to our understanding, but happy to be corrected. @dannyjacobs ?

david-deboer, 2022-08-02:
Maybe just the log_time_stop, since they should always be within a second of each other.  Agree we don't need the timestamp, but it is nice to have in redis.   UTC

david-deboer, 2022-08-02:
I don't love my field names of working, unconfig, maxout, if folks have thoughts.  We should be extensible to other diagnostics, like actually_transmitting

bhazelton, 2022-08-02:
What should this table be called? `snap_configure_state`? @dannyjacobs

david-deboer, 2022-08-02:
`snap_feng_init_status` ?

david-deboer, 2022-08-02:
Incorporate proc_snap_log.py and snap_log.py into hera_corr_f repo.

PR #616 (2022-08-04): Add snap_feng_init_status table

## Description

Add a table to track the status of SNAPs after `feng init`.

## Motivation and Context

closes #615 

## Types of changes

- [ ] Bug fix (non-breaking change which fixes an issue)
- [x] Schema change (any change to the SQL tables)
- [ ] New feature without schema change (non-breaking change which adds functionality)
- [x] Change associated with a change in redis structure
- [ ] Breaking change (fix or feature that would cause existing functionality to change)
- [ ] Version change
- [ ] Build or continuous integration change
- [ ] Other

## Check [... cut]
closed 2022-08-04 by bhazelton