HERA-Team/hera_mc #619: When I2C logging is paused, bogus new values appear in the db

dannyjacobs, 2022-08-10:
Solution:  M&C Correlator monitor should check I2C sensor value timestamps. When the timestamp is unchanged don't put a new value in the db table.
Potential issue: Is there a different timestamp for each sensor value? 

Happy Birthday Bryna!

bhazelton, 2022-08-10:
M&C does monitor the timestamps in redis and only puts a new value in when there's a new timestamp. This is caused by some of the data in a redis key being updated, including the timestamp, but some of the data being persisted rather than nulled. The real solution needs to come upstream, I thought @david-deboer was going to make an issue on hera_corr_f for this but I don't see it.

bhazelton, 2022-08-10:
Here's the hera_corr_f issue: https://github.com/HERA-Team/hera_corr_f/issues/113

david-deboer, 2022-08-11:
I've updated hera_corr_f to provide `json.dumps(None)` when the i2c monitoring is off (this is the same as used for other non-responsive fields in the script).  I also exposed the `include_i2c` flag to the script, so that it can be run with i2c on from the command line.  It is a PR awaiting review.

david-deboer, 2022-08-12:
This changed has been merged and installed on site.

closed 2022-08-12 by david-deboer