HERA-Team/hera_mc #625: monitor correlator catcher most recent file

dannyjacobs, 2022-09-07:
meeting with @AaronParsons and @plaplant they request observers to check the current catcher status and the current file being written. We believe the catcher status to be correctly reporting out to the "Currently Recording" box, but not the file. This issue captures progress towards logging the catcher file writing.

Currently recording box on heranow.
[image]

As of today, the correlator catcher thread is now reporting its most recent file written.   @plaplant reports "the catcher thread is now writing to a new redis key corr:current_file. there is a filename field and a time field. I have installed the latest version of catcher_revamp branch on site"

@bhazelton if you can get this going to the catcher M&C table I will plumb it out to the dashboard

mkolopanis, 2022-09-07:
Not to detract from getting this info in to M&C, but we can add this to the little system summary box you've highlighted already. it is using grafana to make queries straight from redis.

dannyjacobs, 2022-09-07:
Yes. Thats the idea. Other dashboards too.

bhazelton, 2022-09-07:
@dannyjacobs it seems to me like this needs to be in a new table, it doesn't readily fit into the existing tables. The only one I could see working it into is `correlator_component_event_time` but then I'd need to add a column for the filename.

What are your thoughts?

PR #627 (2022-09-09): add correlator_catcher_file table

## Description

Add a table to track the files written by the catcher.

Still need to:
- [ ] update the mc_definitions pdf

## Motivation and Context

closes #625

## Types of changes

- [ ] Bug fix (non-breaking change which fixes an issue)
- [x] Schema change (any change to the SQL tables)
- [ ] New feature without schema change (non-breaking change which adds functionality)
- [ ] Change associated with a change in redis structure
- [ ] Breaking change (fix or feature that would cause existing functionality to change)
- [ ] Version change
- [ ] Build or con [... cut]
closed 2022-09-09 by bhazelton